MangoAutomation/ma-core-public #1679: Log4j2 Rollover Strategy Review

terrypacker, 2020-12-15:
Our default rollover strategy for ma.log is to keep 7 logs per day of a size of 5MB each.  The total number of logs kept is 20.  For most systems this is probably fine but in larger systems generating lots of logs this results is lost logs.  

I suggest we consider adding the following:

1.  No maximum number of logs per day
2. 7 days of logs
3. No more than 1GB of logs (or some other size)

Here is the modified XML config

```
<RollingRandomAccessFile name="logfile" filePattern="${sys:ma.logs}/%d{MM-dd-yyyy}-%i.ma.log" fileName="${sys:ma.logs}/ma.log">
			<PatternLayout pattern="%-5p %d{ISO8601} (%C.%M:%L) - %m%n"/>
			<Policies>
				<OnStartupTriggeringPolicy />
				<TimeBasedTriggeringPolicy />
				<SizeBasedTriggeringPolicy size="5MB"/>
			</Policies>
			<DefaultRolloverStrategy fileIndex="nomax">
				<Delete basePath="${sys:ma.logs}" >
					<IfFileName regex=".*.ma.log" />
					<IfLastModified age="7d" />
          <IfAccumulatedFileSize exeeds="1 GB"/>
				</Delete>
			</DefaultRolloverStrategy>
</RollingRandomAccessFile>
```

jazdw, 2020-12-29:
@terrypacker You can now use env.properties in the default log4j2.xml file so we will probably want to use these to make the number of log files / size easily configurable. e.g. `size="${mango:appender.logfile.size}"`

jazdw, 2020-12-30:
Implemented configurable age, count and size strategies. If any of the conditions match then log files are deleted.

closed 2020-12-30 by jazdw