MangoAutomation/ma-core-public #1688: NoSuchFileException from DiskUsageMonitoringService

jazdw, 2021-01-13:
Seeing a lot of these in tests and presumably on systems without the NoSQL module installed. We should get retrieve an instance of NoSQLProxy (if one exists and Mango is configured to use one) and get the directory from the instance, not a static method.

```
ERROR 2021-01-12T15:34:21,017 (com.infiniteautomation.mango.spring.components.DiskUsageMonitoringService.doPoll:211) - Unable to get NoSQL partition usage 
java.nio.file.NoSuchFileException: /tmp/MangoTestBase1311841488105966210/databases
	at sun.nio.fs.UnixException.translateToIOException(UnixException.java:92) ~[?:?]
	at sun.nio.fs.UnixException.rethrowAsIOException(UnixException.java:111) ~[?:?]
	at sun.nio.fs.UnixException.rethrowAsIOException(UnixException.java:116) ~[?:?]
	at sun.nio.fs.UnixFileStore.devFor(UnixFileStore.java:59) ~[?:?]
	at sun.nio.fs.UnixFileStore.<init>(UnixFileStore.java:66) ~[?:?]
	at sun.nio.fs.LinuxFileStore.<init>(LinuxFileStore.java:49) ~[?:?]
	at sun.nio.fs.LinuxFileSystemProvider.getFileStore(LinuxFileSystemProvider.java:51) ~[?:?]
	at sun.nio.fs.LinuxFileSystemProvider.getFileStore(LinuxFileSystemProvider.java:39) ~[?:?]
	at sun.nio.fs.UnixFileSystemProvider.getFileStore(UnixFileSystemProvider.java:373) ~[?:?]
	at java.nio.file.Files.getFileStore(Files.java:1487) ~[?:?]
	at com.infiniteautomation.mango.spring.components.DiskUsageMonitoringService.doPoll(DiskUsageMonitoringService.java:206) ~[classes/:?]
	at org.springframework.security.concurrent.DelegatingSecurityContextRunnable.run(DelegatingSecurityContextRunnable.java:82) ~[spring-security-core-5.4.2.jar:5.4.2]
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128) ~[?:?]
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628) ~[?:?]
	at java.lang.Thread.run(Thread.java:834) [?:?]
```

jazdw, 2021-02-08:
@terrypacker is this complete then?

terrypacker, 2021-02-08:
@jazdw I haven't seen the messages since then.  Calling this done.

closed 2021-02-08 by terrypacker