MangoAutomation/ma-core-public #1708: Clean/Purge Orphaned DataPointVO.seriesId

terrypacker, 2021-03-09:
The nightly data purge removes point values for orphaned series Ids but does nothing to maintain the timeSeries table.

- [ ] Check to see if seriesId is removed when deleting a data point
- [ ] Potentially ensure no orphaned series Ids are in timeSeries table during purge.

terrypacker, 2021-04-09:
The DataPointDao does delete the seriesId entry if possible when deleting post relational data (also in the bulk delete method).  So unless we feel the orphaned seriesId check is necessary during the nightly purge we can close this.

terrypacker, 2021-06-09:
@jazdw any opinion here?  Otherwise we can close this.

jazdw, 2021-06-09:
@terrypacker should be done at some point. Don't just close.

terrypacker, 2021-06-09:
Moved to Jira.  https://radixiot.atlassian.net/browse/RAD-948

closed 2021-06-09 by terrypacker