MangoAutomation/ma-core-public #1720: Purging All Point Values Does Not Purge the MapDB cache

terrypacker, 2021-03-31:
Test was done with data source disabled.  I can see in the logic that this is the problem.  Likely fix:

RuntimeManagerImpl ~778

 ```    private void updateDataPointValuesRT(int dataPointId, long before) {
        DataPointRT dataPoint = dataPoints.get(dataPointId);
        if (dataPoint != null)
            // Enabled. Reset the point's cache.
            dataPoint.resetValues(before);
    }
```
We will need some logic similar to the PointValueCache.reset(...) methods that will use the PointValueCacheDao to reset the stored cache after the purge.

Probably need to consider performance as this will happen in the nightly build and very usually NOT purge the latest values.

terrypacker, 2021-04-01:
@jazdw please take a look at this fix when you get a chance.

jazdw, 2021-04-06:
@terrypacker The deleteCachedValue etc methods don't look thread safe to me

terrypacker, 2021-04-09:
Also I saw this in the nightly purge log:

```
ERROR 2021-04-09T04:00:52,187 (com.serotonin.m2m2.rt.maint.DataPurge$DataPurgeTask.run:345) - null
java.lang.NullPointerException: null
	at com.infiniteautomation.mango.mapdb.MapDbPointValueCacheDao.deleteCachedValuesBefore(MapDbPointValueCacheDao.java:169) ~[?:?]
	at com.serotonin.m2m2.rt.RuntimeManagerImpl.purgeDataPointValuesWithoutCount(RuntimeManagerImpl.java:772) ~[classes/:?]
	at jdk.internal.reflect.GeneratedMethodAccessor445.invoke(Unknown Source) ~[?:?]
	at jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43) ~[?:?]
	at java.lang.reflect.Method.invoke(Method.java:566) ~[?:?]
	at com.infiniteautomation.mango.spring.components.RunAsImpl.lambda$runAsProxy$0(RunAsImpl.java:89) ~[classes/:?]
	at com.sun.proxy.$Proxy69.purgeDataPointValuesWithoutCount(Unknown Source) ~[?:?]
	at com.serotonin.m2m2.rt.maint.DataPurge.purgePoint(DataPurge.java:171) ~[classes/:?]
	at com.serotonin.m2m2.rt.maint.DataPurge.executeImpl(DataPurge.java:83) ~[classes/:?]
	at com.serotonin.m2m2.rt.maint.DataPurge.execute(DataPurge.java:65) ~[classes/:?]
	at com.serotonin.m2m2.rt.maint.DataPurge$DataPurgeTask.run(DataPurge.java:343) ~[classes/:?]
	at com.serotonin.timer.Task.runTask(Task.java:183) ~[classes/:?]
	at com.serotonin.timer.TaskWrapper.run(TaskWrapper.java:23) ~[classes/:?]
	at com.serotonin.timer.OrderedThreadPoolExecutor$OrderedTaskCollection.run(OrderedThreadPoolExecutor.java:313) ~[classes/:?]
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128) ~[?:?]
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628) ~[?:?]
	at java.lang.Thread.run(Thread.java:834) [?:?]
```

terrypacker, 2021-04-12:
@jazdw when you say "The deleteCachedValue etc methods don't look thread safe to me" I suspect you didn't realize the each individual cache list is read only by design.  So to modify the list a copy is always made and returned.  Am I missing something?

jazdw, 2021-04-13:
While you are looping over the list of PointValueTime the DS may be polling for example and add a new point value to the cache which will then be lost if you call `updatePointValueCache()` no?

terrypacker, 2021-04-13:
True.  The original issue is that disabled points don't get their caches purged.  So I think the solution is to check if the point is running and if so use the original logic of the RuntimeManager to purge it but if it is disabled use the PointValueCacheDao's methods.

terrypacker, 2021-04-13:
I changed the logic in the last commit and the only methods that will have a performance impact are the 2 methods for purging all point data which now will query for all data points and process the caches 1 by 1

terrypacker, 2021-06-09:
Closing as released already.

closed 2021-06-09 by terrypacker